henryk/openmili #4: Association mode

fab33, 2015-07-16:
I try association but it doesn't work. First I start admin (https://github.com/riptidewave93/RFLED-Server/blob/master/source/admin.py) and I can use Milight 2.0 app on Android.
I try new association, Led bulb will flash 3 times (so seems to be ok) but after that I can't switch to white only RGB work (OFF and ON commands doesn't work too).
And after a while, my bulb doesn't work with remote and I can't remove association or associate with the wifi bridge ... :(

fab33, 2015-07-16:
Sorry, with the reset procedure I recover my bulb. But I confirm when I make associating procédure with my Arduino only color work.

fab33, 2015-07-16:
The protocol (old one ?) is at the end of these page : http://www.limitlessled.com/dev/

fab33, 2015-07-16:
So my request : can you improve associate mode. So we don't have to buy the wifi bridge but only bulbs (and more bulbs !)

fab33, 2015-07-16:
One precision : Specials modes (DISCO, ..) doesn't work too (switch to OFF when trying it).

VincentGijsen, 2015-12-29:
At my MSP430 setup, without modification (other than put char), it works great with my RGB+W bullp.s

I use the iPhone Limitless app, which enables white mode ONLY when keeping the power-on button pressed for a longer tie (e.g. 2 sec). the NIGHTMODE is actived by keeping the off button pressed. 

Hope this helps.

fab33, 2015-12-29:
Thank you I will try it

VincentGijsen, 2015-12-30:
Hi @fab33 ,

i'm afraid I have to revert my claim that everything works, I too have issues, but this is with the switching off of groups, see issue #7. Guess there is some timing problems.

Perhaps i'll review the details work of @henryk, and grab some RTL-sdr, although i'm not sure if it has the correct tuner installed to sample at 2.4ghz