henryk/openmili #5: Can't compile in arduino

andrezibaia, 2015-09-22:
Hi,

First things first: thank you for this project! I have been having some trouble getting the milight bridge to work consistently and I hope to overcome its limitations with openmili.

That said, I am having some trouble getting this code to compile in Arduino IDE. I am trying to flash it to an ESP8266 ESP-201 module, but keep hitting the same walls:

I get this error:
"printf.h:34:27: error: 'fdevopen' was not declared in this scope
 fdevopen( &serial_putc, 0 );"

...and if I comment said line, I get this one:

several times "openmili.ino:23: undefined reference to `putchar'" and then "collect2.exe: error: ld returned 1 exit status. Error compiling."

What am I doing wrong?

Thanks!

André

VincentGijsen, 2015-12-29:
Hi,

I've had a similar issue, although i'm using this marvellous code with an MSP430, so ENRGIA, not arduino.

That being said, I had the same issue with put char.

i've added 

```
#undef putchar
int putchar(int c) {
    Serial.write(c);
    return (c);
}
```

into the libraries/RF24/RF24.cpp, although It might work within your own sketch as well.

good luck

andrezibaia, 2015-12-30:
Thank you very much! I will try that and will let you know how it goes.