henryk/openmili #6: Control with arduino

Phil1988, 2015-11-24:
Could someone please tell me how I could control the bulbs with an Arduino and a nRF2401+ module?
I can only see this on every start of the arduino:
"# OpenMiLight Receiver/Transmitter starting"

How could I for example pair the "arduino remote" with the bulb as group2?
I think I have to send some thing like this:
07 B0 F2 EA 7D B0 15 F0

But how do I enter it in the SerialMonitor to send it?
Any help is highly appreciated!

VincentGijsen, 2015-12-29:
well, start of by checking that you have indeed the proper wiring setup. 

After which, don't fiddle around on the Serial Monitor.
Instead: execute the python script on you computer, and point it tho the serial port where the arduino is available.
When you have that running, grab an app for your smartphone/pc/whatever and configure it to either broadcast or directly send it's UDP traffic to your pythons app (per default it listens on 8899).

if this works, than reverse engineer the Python script, as it mangles the UDP packages, prior to sending it to the aruino/msp430.  The script will explain how to write the bytes into the serial buffer.  You might like to improve the arduino side of things, to directly parse the packages (in which case, please share your modifications with the community!

Phil1988, 2015-12-30:
@VincentGijsen: Thanks you a lot for answering!

I am sure that my wireing is right (I also wrote a simple "chat program" between my two nRF24L01 modules to check if they can communicate and they do).
 
Sadly I'm not familiar with python (and I am using Win7).

I installed Python 3.5.1 on windows and run cmd after this.
When I now run the "openmilid.py" I get this error:
__________________
....\host\openmilid.py", line 109
    print fd.read(1024)
           ^
SyntaxError: invalid syntax
__________________

So I dont know how to execute the python script neither how to point it to the serial port of my arduino (COM 2).

You see I am already lost at the basics.
Could you explain this more to me or do you have a video how to do this?

I thank you a lot!

VincentGijsen, 2015-12-30:
Hi,

Googleing, the first hit on ```SyntaxError: invalid syntax``` yields the following: 
http://stackoverflow.com/questions/7584489/python-print-statement-syntax-error-invalid-syntax

This means your are using python 3 instead of two. which is basally ok, although ```print```is a function, and functions are called using the parentheses so **( stuff ).**

Just modify the line ```print fd(1024) ``` into ```print(fd.read(1024))```

I guess, after fixing this (and possibly other print statements), you may get an new exception that *serial* is unavailable.

You first need to install ```pip```, which may or may not be available with your current python install, if not, google for **python install pip**

after which you may execute in your **cmd** box:
```pip install pyserial```

*please note, i'm no python programmer whatsoever, google is occasionally my best friend*

Phil1988, 2015-12-30:
Also if you are not a python programmer - your knowledge is far beyond mine :-)
this solved the problem and put me into new ones...

_______________
Traceback (most recent call last):
...openmilid_modified.py", line 129, in <module>
    OpenMiLiDaemon( args ).start()
...openmilid_modified.py", line 78, in start
    with serial.Serial(self.args.serial, self.args.baudrate, timeout=0.1) as fd:
...\Python35-32\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    SerialBase.__init__(self, *args, **kwargs)
...\Python35-32\lib\site-packages\serial\serialutil.py", line 171, in __init__
    self.open()
...\Python35-32\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port %r: %r" % (self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/dev/ttyUSB0': FileNotFoundError(2, 'Das System kann den angegebenen Pfad nicht finden.', None, 3)
_______________

The only thing I see here is wrong is the port direction as it is linux like as /dev/.. and should be something like this:

_______________
import serial
ser = 0
def init_serial():
    portnumber = 2        
    global ser        
    ser = serial.Serial()
    ser.baudrate = 9600
    ser.port = portnumber 
_______________

But as you see... I am completely lost with this...

VincentGijsen, 2015-12-30:
Hi Phil,

when you look at the last lines of the Host app:

```
<<SNIP>>
 parser = argparse.ArgumentParser(description='OpenMiLi daemon -- Listen for commands on the network and control Mi-Light bulbs.')
    parser.add_argument('--serial', '-s', help="Device node for the serial line to connect through", default="/dev/ttyUSB0")
    parser.add_argument('--baudrate', '-b', help="Baud rate for the serial line", type=int, default=115200)
```
You will notice that this is a parser for the command line options. When you run the script with ``--serial <your port>```, it will try to connect to the port.

The default is /dev/ttyUSB0, which refers to the first USB serial adapter in linux, for Windows, however you'll need to substitute this for COM1 / COM3, or whatever. Just consult your **Device manager in Windows** to check on with port your Arduino or serial adapter is registered. Probably you could try to use the same port the Arduino IDE uses.

Also ensure that the `` baudrate ``` (which defaults to 115200) is the same, both for arduino, as for the python app.

Should this still not work, you may uninstall Python 3, and install Python 2.7.

Than install pip, and try the unmodified version of the python app (so revert any changes for the print statements)

rremigius, 2019-10-15:
Hi, I know I'm replying to an age-old Issue, but it's still open and I was hoping to find an answer here as well.

I just started out, got 2 Arduino's to message each other with RF24L01+ modules to verify I got the wiring right, and now I'm trying to get some packets on the screen - without any luck so far. The sketch is running on Arduino, the openmilid.py script is running and shows 

```
# OpenMiLight Receiver/Transmitter starting
```

And now I don't know what I'm supposed to do. Nothing I do with my lamps or controller shows any feedback on the screen. I installed some Milight apps, pointed them at my computer and started pressing and swiping some colors. The most I got there was:

```
# Error: outgoing packet complete. Press enter to send.
# Error: outgoing packet complete. Press enter to send.
```

Even if they did show something, I wouldn't know how to use that information to control my lights from a script. Probably I misunderstood how this library is supposed to work, but I can't get it from the readmes or code. Are there perhaps any sort of "Hello World" instructions that get you to your very first command-line light switch?

VincentGijsen, 2019-10-16:
If I recall correctly, you have to pair each bulb to the address
configured. And your sync word (from recollection) cannot be everything.
Withe any byte value.

Was there also a sniffer example? Otherwise you can use your remote to
receive it's messages. The bulbs only listen , and don't send anything.